JsBergbau/MiTemperature2 #88: ATC mode using call reporting outdated values

thepiwo, 2021-05-28:
using this command:

`/home/pi/MiTemperature2/LYWSD03MMC.py --atc --watchdogtimer 5 --battery --round -deb -call ./prometheus/sendToPrometheus.py --devicelistfile /home/pi/MiTemperature2/devicefile.ini --onlydevicelist`

Reported values in call seems to be stuck over time and what's send doesn't change in a significant way. I fail to find where the actual values that are sent are gathered.

Before a restart of the script the values only change in minor way. After restart the values change instantly to the actually broadcasted values. The Android MijiaTemp App always reports the correct latest values, as does the BLE packet log.

e.g.
```
May 28 12:02:02 raspberrypi1 python3[2073]: BLE packet: A4:C1:38:1F:BF:C8 00 1110161a18a4c1381fbfc800cc29300a4fd9 -63
May 28 12:02:02 raspberrypi1 python3[2073]: Temperature:  20.4
May 28 12:02:02 raspberrypi1 python3[2073]: Humidity:  41
May 28 12:02:02 raspberrypi1 python3[2073]: Battery voltage: 2.639 V
May 28 12:02:02 raspberrypi1 python3[2073]: RSSI: -63 dBm
May 28 12:02:02 raspberrypi1 python3[2073]: Battery: 48 %
May 28 12:02:02 raspberrypi1 python3[2073]: /home/pi/MiTemperature2/prometheus/sendToPrometheus.py sensorname,temperature,humidity,voltage,batteryLevel,timestamp temperature-desk 20.5 40 2.637 48 1622196020
...
May 28 12:03:00 raspberrypi1 python3[2073]: BLE packet: A4:C1:38:1F:BF:C8 00 1110161a18a4c1381fbfc800ca29300a4ae0 -74
May 28 12:03:00 raspberrypi1 python3[2073]: Temperature:  20.2
May 28 12:03:00 raspberrypi1 python3[2073]: Humidity:  41
May 28 12:03:00 raspberrypi1 python3[2073]: Battery voltage: 2.634 V
May 28 12:03:00 raspberrypi1 python3[2073]: RSSI: -74 dBm
May 28 12:03:00 raspberrypi1 python3[2073]: Battery: 48 %
May 28 12:03:00 raspberrypi1 python3[2073]: /home/pi/MiTemperature2/prometheus/sendToPrometheus.py sensorname,temperature,humidity,voltage,batteryLevel,timestamp temperature-desk 20.5 41 2.637 48 1622196060
...
May 28 12:03:51 raspberrypi1 python3[2073]: BLE packet: A4:C1:38:1F:BF:C8 00 1110161a18a4c1381fbfc800ca29310a52e5 -68
May 28 12:03:51 raspberrypi1 python3[2073]: Temperature:  20.2
May 28 12:03:51 raspberrypi1 python3[2073]: Humidity:  41
May 28 12:03:51 raspberrypi1 python3[2073]: Battery voltage: 2.642 V
May 28 12:03:51 raspberrypi1 python3[2073]: RSSI: -68 dBm
May 28 12:03:51 raspberrypi1 python3[2073]: Battery: 49 %
May 28 12:03:51 raspberrypi1 python3[2073]: /home/pi/MiTemperature2/prometheus/sendToPrometheus.py sensorname,temperature,humidity,voltage,batteryLevel,timestamp temperature-desk 20.4 41 2.637 48 1622196090
```

JsBergbau, 2021-05-28:
Hi thepiwo,
for debugging this issue, please use `sendToFile.sh` as callback. Then in another window do `tail -f data.txt` and have a look if there are the current values which are also shown on the displays of the sensor.

thepiwo, 2021-05-28:
Indeed logging to file seems to be current all the time with the display on the sensor, in the data file as well as the python log.

Does that mean the pushgateway is just to slow and getting stuck in some queue?

thepiwo, 2021-05-28:
Actually it seems like exactly this, the sendToPrometheus script is always at 100% CPU. Can I limit the interval of the sending some way?

JsBergbau, 2021-05-28:
What device are you running on? How many sensors do you use?

> Can I limit the interval of the sending some way?

The easiest way is  to change these parameters
[image]
2500ms x 4 = 10.000ms, this means every 10 seconds a new value. So increase these parameters for having less frequent new data transmitted via callback.

thepiwo, 2021-05-28:
It's the LYWSD03MMC, with 11 devices total, 5 on one pi zero, 6 on the other.

Thanks, I guess I'll look for a solution where I integrate the sendToPrometheus in the main script on order to not have the full python initialization for each value to be sent.

JsBergbau, 2021-05-28:
I don't have experience with Prometheus, but perhaps it is also possible to get the data via MQTT into it. Or you have to look into the script and change it that way that the prometheus code is integrated. I would also then check if prometheus accepted that data, so that it is retransmitted when prometheus was offline and goes back online and no data is lost.

thepiwo, 2021-05-28:
Thank you so far, I will try out a few things.

thepiwo, 2021-05-28:
indeed initializing python for each callback was the bottleneck, I copied the prometheus push to gateway over the default callback and it works without issue and without increased cpu load.

https://github.com/thepiwo/MiTemperature2/commit/adce55bc89cd84ac81773249f7d4c2b59f317be9

closed 2021-05-28 by thepiwo