JsBergbau/MiTemperature2 #89: Read values in ATC mode failed

sl4m01, 2021-06-02:
Dear all,
i do have 3 LYWSD03MMC sensor.
On 1 i flashed the custom firmware (https://github.com/pvvx/ATC_MiThermometer#bluetooth-advertising-formats). 3 sensors do have stock firmware.
Now i tried to read values from the sensors using my RPI 3b+

The stock firmware works fine:
`python3 LYWSD03MMC.py -d A4:C1:38:AD:3E:FA -b
---------------------------------------------
MiTemperature2 / ATC Thermometer version 3.1
---------------------------------------------
Trying to connect to A4:C1:38:AD:3E:FA
Temperature: 24.84
Humidity: 20
Battery voltage: 2.903 V
Battery level: 80
`

However reading from the custom firmware doesnt work:
`python3 LYWSD03MMC.py -d A4:C1:38:E6:FF:3E -b -a
---------------------------------------------
MiTemperature2 / ATC Thermometer version 3.1
---------------------------------------------
Trying to connect to A4:C1:38:E6:FF:3E
Connection lost
Waiting...
Trying to connect to A4:C1:38:E6:FF:3E
Connection lost
Waiting...
Trying to connect to A4:C1:38:E6:FF:3E`

Both sensors can be found by the RPI3b+ using the sudo hcitool lescan (the are located just next to the RPI3b+).

I also tried the suggested method in the readme:
`python3 LYWSD03MMC.py --atc --mqttconfigfile mqtt.conf --devicelistfile sensors.ini
---------------------------------------------
MiTemperature2 / ATC Thermometer version 3.1
---------------------------------------------
Script started in ATC Mode
----------------------------
In this mode all devices within reach are read out, unless a devicelistfile and --onlydevicelist is specified.
Also --name Argument is ignored, if you require names, please use --devicelistfile.
In this mode rounding and debouncing are not available, since ATC firmware sends out only one decimal place.
ATC mode usually requires root rights. If you want to use it with normal user rights,
please execute "sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f `which python3`)"
You have to redo this step if you upgrade your python version.
----------------------------
Power ON bluetooth device 0
Bluetooth device 0 is already enabled
Enable LE scan
scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no
socket filter set to ptype=HCI_EVENT_PKT event=LE_META_EVENT
Listening ...
MQTT connected with result code 0`

Also this stays like this for a long time.

sensors.ini
`
[info]
;info1=now all available options are listet. If offset1, offset2, calpoint1 and calpoint2 are given 2Point calibration is used inste$
;info1= Note options are case sensitive
;info4=Use semicolon to comment out lines
;sensorname=Specify an easy readable name
;humidityOffset=-5
;offset1 = -10
;offset2 = 10
;calpoint1 = 33
;calpoint2 = 75
;topic=This sensor data will be published with this name when using integrated MQTT
;If no topic is given, then default topic from MQTT config is used

;[A4:C1:38:DD:EE:FF]
; sensorname=Bathroom
; humidityOffset=-30
; offset1 = -10
; offset2 = -10
; calpoint1 = 33
; calpoint2 = 75
; topic = basement/Bathroom

[A4:C1:38:E6:FF:3E]
sensorname=Living Room
topic=basement/Living_Room`

Can someone help or give a hint?

Attached scrrenshots of the custom FW config
[image]
[image]

Best regards, Simon

JsBergbau, 2021-06-02:
Hi Simon,

try `python3 LYWSD03MMC.py --atc  --watchdogtimer 5`
Do you now see your ATC flashed sensor?

Important: Make sure you aren't connected anymore to ATC sensor. As long as there is the bluetooth symbol on display no advertisments are sent that the script could decode.

sl4m01, 2021-06-02:
Dear JsBergbau,
thank you for your answer!
Unfortunatelly no improvement.
However, i asume i have to put -d MAC address to the command?

Also there is no bluetooth symbol in the display.
However, we i run the script its shows up and disappears after a few seconds.

```
python3 LYWSD03MMC.py -d A4:C1:38:E6:FF:3E --atc --watchdogtimer 5
---------------------------------------------
MiTemperature2 / ATC Thermometer version 3.1
---------------------------------------------
Trying to connect to A4:C1:38:E6:FF:3E
Connection lost
Waiting...
Trying to connect to A4:C1:38:E6:FF:3E
Connection lost
Waiting...
Trying to connect to A4:C1:38:E6:FF:3E
Connection lost
Waiting...
Trying to connect to A4:C1:38:E6:FF:3E
Connection lost
Waiting...
Trying to connect to A4:C1:38:E6:FF:3E
Connection lost
Waiting...
Trying to connect to A4:C1:38:E6:FF:3E
Connection lost
Waiting...
Trying to connect to A4:C1:38:E6:FF:3E
Connection lost
```

sl4m01, 2021-06-02:
Sorry, i was to fast!
You are right!
In case i dont put the MAC address the following shows up:
```
python3 LYWSD03MMC.py --atc --watchdogtimer 5
---------------------------------------------
MiTemperature2 / ATC Thermometer version 3.1
---------------------------------------------
Script started in ATC Mode
----------------------------
In this mode all devices within reach are read out, unless a devicelistfile and --onlydevicelist is specified.
Also --name Argument is ignored, if you require names, please use --devicelistfile.
In this mode rounding and debouncing are not available, since ATC firmware sends out only one decimal place.
ATC mode usually requires root rights. If you want to use it with normal user rights,
please execute "sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f `which python3`)"
You have to redo this step if you upgrade your python version.
----------------------------
Power ON bluetooth device 0
Bluetooth device 0 is already enabled
Enable LE scan
scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no
socket filter set to ptype=HCI_EVENT_PKT event=LE_META_EVENT
Listening ...
Watchdog: Did not receive any BLE Paket within 1622648487 s. Restarting BLE scan. Count: 1
Disable LE scan
Enable LE scan
scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no

BLE packet: A4:C1:38:E6:FF:3E 00 1110161a18a4c138e6ff3e011b2a590bbc55 -63
Temperature:  28.3
Humidity:  42
Battery voltage: 3.004 V
RSSI: -63 dBm
Battery: 89 %

BLE packet: A4:C1:38:E6:FF:3E 00 1110161a18a4c138e6ff3e011b2a590bba56 -68
Temperature:  28.3
Humidity:  42
Battery voltage: 3.002 V
RSSI: -68 dBm
Battery: 89 %

BLE packet: A4:C1:38:E6:FF:3E 00 1110161a18a4c138e6ff3e011b2a590bba57 -53
Temperature:  28.3
Humidity:  42
Battery voltage: 3.002 V
RSSI: -53 dBm
Battery: 89 %
```

Is this correct? :D 
Is this now also published in mqtt?
Can you give me a hint how to autostart this script on the RPI on bootup?

JsBergbau, 2021-06-02:
Create a fill like "startup.sh" in MiTemperature2 folder and put into it
```bash
#!/bin/bash
DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" >/dev/null 2>&1 && pwd )"
cd $DIR
 python3 LYWSD03MMC.py --atc --mqttconfigfile mqtt.conf --watchdogtimer 5 --devicelistfile sensors.ini
```

then exec `crontab -e` and add a line `@reboot <path to your script> >/dev/null`

sl4m01, 2021-06-03:
Hi JsBergbau,
thank you alot!
Because i am really a raspberry pi and linux (?) noob i only managed to get to the following point:
In case i save the following line in `crontab -e`
`40 18 * * * /home/openhabian/MiTemperature2/Xiaomi_sensor_startup.sh >/dev/null`
I am receiving mqtt signals from 18:40 on.
However, in case save the following: 
`@reboot /home/openhabian/MiTemperature2/Xiaomi_sensor_startup.sh >/dev/null`
Nothing happens after reboot (no mqtt values) ...
this is really strange. Do you have an idea?

closed 2021-06-04 by sl4m01