JsBergbau/MiTemperature2 #92: Attaching a DS18B20 probe Data wire location

Sukalo0290, 2021-06-23:
I plan on adding one of these water proof probes to replace the sensor but i'm wondering where the data wire would be attached.

[image]

I plan on editing this file 
https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer/sensor.c

Where would i attach the data wire? I have seen the diagram but where would the data cable go?

Red connects to 3-5V - already know
Blue/Black connects to ground - already know
Yellow/White is data - not sure

[image]

JsBergbau, 2021-06-23:
Hi Sukalo0290,
thanks for your interest. However I'm just developing a script to read transmitted data. So I can't help you to connect the DS18B20 to LYWSD03MMC Temperature sensor. I don't even know whether firmware can read the data or not. 
I see you have already asked it there https://github.com/pvvx/ATC_MiThermometer/issues/105 
So I'm closing here.

closed 2021-06-23 by JsBergbau

Sukalo0290, 2021-06-23:
No problem, i didn't know if these questions would reach others who might be able to help. But thanks for all your work already, script all working great.