JsBergbau/MiTemperature2 #93: Encrypted advertisement - ATC mode

rocket59, 2021-06-24:
I have successfully set up one of these devices and am able to read out the data on my little pi zero, thanks for the work you did on this @JsBergbau ! 

I was wondering however if there is/would be support for decrypting the bindkey encrypted advertisement option in the ATC firmware?

JsBergbau, 2021-06-25:
Hi rocket59,
indeed I'd love to have support for encrypted ATC mode. In fact I've asked the developer to implement it. In addition I've added an option in the Telink-Flasher Webpage to generate a MiBind Key via a password, because I find that much more practical than handling this long "cryptic" key. Currently I'm wating for the developer to integrate this into its Flasher https://github.com/pvvx/pvvx.github.io/pull/3 
When it is integrated I'll add support for encrypted ATC mode - with and without password. I'll ask again what is missing to get it integrated.

rocket59, 2021-06-25:
Fantastic! Look forward to seeing this added!

rocket59, 2021-07-10:
Hi @JsBergbau, it seems that the dev appears reluctant to add the option due to licencing concerns? If so I'd be happy to use the stock bind key or even generate the key online from the argon2 site despite being more cumbersome. In any case it's a one off effort for each sensor so perhaps not necessarily a huge burden.

JsBergbau, 2021-07-10:
To be honest, I didn't understand what is currently preventing him to merge the pull request, because other people commented, that it is allowed with the license to use the code. Can you ask at the issue for an update? Perhaps pvvx will integrate it then.

rocket59, 2021-07-30:
Hi @JsBergbau, looks like pvvx hasn't been active this month, hopefully he's doing ok. Any chance of getting an update to the code as per my last post? This would significantly improve security of the thermometer transmissions!

JsBergbau, 2021-07-30:
I'll definitively implement it. Lets wait about another another month for a reply of pvvx. Currently I have a lot to do, so in another month I hope to have more sparetime to implement.

rocket59, 2021-07-30:
No problem - thanks!

rocket59, 2021-09-16:
Hi @JsBergbau was wondering if you've had a chance to revisit this? I'm thinking pvvx is probably unlikely to update the flasher for password support.

rocket59, 2021-11-13:
Sorry life got in the way !

So I'm just bit unclear as to how to run in ATC mode. If I can list the additional steps I'm taking, please correct me if I'm doing anything wrong:

1. install pybluez and pycryptodomex per the prerequisites
2. flash the sensor with the new pvvx 3.4 firmware 
3. set telinkflasher options per https://github.com/JsBergbau/MiTemperature2#encrypted-atc-mode and create the devicelist file. I note that telinkflasher has not been updated to generate the bind key from a password so I assume that we need to generate the key ourselves using argon as per https://github.com/pvvx/ATC_MiThermometer/issues/94#issuecomment-847142307 (or just create a random 32bit key). 
4. I guess we can just use the same suggested argon options per your original pull request to pvvx here https://github.com/pvvx/pvvx.github.io/pull/3/files? Is it also correct to assume that "time:10" in the pull request is the number of iterations in the argon web page https://antelle.net/argon2-browser/? 
5. The resultant hash output from Argon is the key that we would enter into the devicelist file as well as the into the telinkflasher field titled "EEP Bindkey" and then hit the set button?
6. get the new LYWSD03MMC.py file
7. Run python3 LYWSD03MMC.py --devicelistfile DEVICELISTFILE

Sorry for the lengthy steps and thanks again!

JsBergbau, 2021-11-15:
Hi rocket59,

thanks for your confirming, that it works.

> I'm still a little confused by how the concept of multiple receivers and node-red works (I use node red to receive/send MQTT messages at the moment so mostly get node red).

Basically every receiver is running MitTemperature2. MiTemperature2 reports received values to a certain MQTT topic. In Node-RED there is an MQTT Node which receives this message. Every MQTT message contains a key `receiver` so it is clear where this message came from. Of course this gets the same message for some Thermometers. When using it with influxdb and adjusting the timestamps for example to a 25 seconds interval (that's the interval I'm using) then double values are OK, because influxdb only stores one value for each unique timestamp. Latest message will override previously stored DB entry.

rocket59, 2021-11-15:
Ah, I get it now. I've got couple of pi zero's at opposite ends of my house and I'm splitting the various sensors between them so probably won't be running this scenario in any case. 

Thanks again!

closed 2021-11-15 by rocket59