JsBergbau/MiTemperature2 #102: Warning certificate

ZencoreCZ, 2021-11-01:
error is where please?

/MiTemperature2# python3 LYWSD03MMC.py -d <myMAC> -c 1 --name Exam --httpcallback "https://mydomain.com/myscript?name={sensorname}&temp={temperature}&hum={humidity}&bat={batteryLevel}

`InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings`

JsBergbau, 2021-11-01:
It looks like that any SSL certificate is accepted and thats why the warning is thrown. Hope @mvdklip can explain it more.

mvdklip, 2021-11-05:
@JsBergbau is right. The request is being made with the verify=False flag so that people running their home automation on a self generated certificate can use it. The requests library issues a warning when doing this. It's not an error.

mvdklip, 2021-11-05:
If you really wanted to hide the warning which I don't really recommend you could have a look here:

https://stackoverflow.com/questions/15445981/how-do-i-disable-the-security-certificate-check-in-python-requests

But that requires changes to the Python code.

JsBergbau, 2021-11-05:
Thanks for the info. So it makes sense, that there is a warning displayed.