JsBergbau/MiTemperature2 #104: illegal instruction in ATC mode

lsroka76, 2021-12-12:
Hi,

i've installed everything according to instructions on my Pi Zero W but script crashes with "illegal instruction" when used with --atc. 
There are no issues without --atc switch. I tried everything including reinstallation of Pi system from scratch.

JsBergbau, 2021-12-12:
Hi Isroka76,

thanks for your report. Based on your post this sounds so far either as bug in some python version or maybe even a hardware problem with your Pi Zero W.
Please post detailed steps of how you installed your Pi and MiTemperature2 script as well es the complete commandline running MiTemperature2 script.

lsroka76, 2021-12-13:
I did several attempts, each time starting from scratch (SD format, fresh OS image - Raspberry Pi OS Lite (32-bit), 10.2021 using Raspberry Pi Imager). I tried with "sudo apt-get dist-upgrade" and without it...I even tried to built Bluez from sources.

Nothing helps - script works perfectly with 5 sensors, but with --atc it ends with "illegal instruction" - with and without sudo.

JsBergbau, 2021-12-13:
Since there are some requirements to be installed before script can run, please also post all the steps after you have setup the Raspberry PI OS.

lsroka76, 2021-12-13:
sudo apt-get update

sudo apt-get dist-upgrade

sudo rpi-update

sudo apt-get autoclean

sudo reboot

sudo apt-get install -y mc git libssl-dev bc python3-pip supervisor

pip3 install bluepy requests

sudo apt install bluetooth libbluetooth-dev

pip3 install pybluez

sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f `which python3`)

JsBergbau, 2021-12-14:
It seems to be a problem with pycryptodomex library. 
When you don't want to use encrypted mode, open `cryptoFunctions.py` and comment line `#from Cryptodome.Cipher import AES`

lsroka76, 2021-12-14:
It works! Thank you :-)

closed 2021-12-14 by lsroka76

JsBergbau, 2022-01-22:
Old version is not needed any longer. `pip3 install pycryptodomex` works again.