JsBergbau/MiTemperature2 #106: failed to read broadcasted data in ATC module

SydneyOwl, 2021-12-18:
I tried `python3 LYWSD03MMC.py –atc –watchdogtimer 5` and `python3 LYWSD03MMC.py –atc`, but neither of them can work.  When –watchdogtimer is added, LE Scan keeps stopping and starting. If removed, nothing display on the screen except “Listening……”
```shell
Power ON bluetooth device 0
Bluetooth device 0 is already enabled
Enable LE scan
scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no
socket filter set to ptype=HCI_EVENT_PKT event=LE_META_EVENT
Listening ...
```
```shell
Power ON bluetooth device 0
Bluetooth device 0 is already enabled
Enable LE scan
scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no
socket filter set to ptype=HCI_EVENT_PKT event=LE_META_EVENT
Listening ...
Watchdog: Did not receive any BLE Paket within 1639795077 s. Restarting BLE scan. Count: 1
Disable LE scan
Enable LE scan
scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no

Watchdog: Did not receive any BLE Paket within 1639795083 s. Restarting BLE scan. Count: 2
Disable LE scan
Enable LE scan
scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no

Watchdog: Did not receive any BLE Paket within 1639795089 s. Restarting BLE scan. Count: 3
Disable LE scan
Enable LE scan
scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no
```
but data broadcasted can be seen by running `bluetoothctl` `scan on` on ubuntu 20.04:
```shell
[CHG] Device A4:C1:38:4C:5A:3D RSSI: -28
[CHG] Device A4:C1:38:4C:5A:3D ServiceData Key: 0000181a-0000-1000-8000-00805f9b34fb
[CHG] Device A4:C1:38:4C:5A:3D ServiceData Value:
  a4 c1 38 4c 9d 34 01 0d 35 64 0c 58 bd           ..8L.4..5d.X. 
……
[CHG] Device A4:C1:38:4C:5A:3D RSSI: -26
[CHG] Device A4:C1:38:4C:5A:3D ServiceData Key: 0000181a-0000-1000-8000-00805f9b34fb
[CHG] Device A4:C1:38:4C:5A:3D ServiceData Value:
  a4 c1 38 4c 9d 34 01 0d 35 64 0c 58 bf           ..8L.4..5d.X.  
```
10 secs between each broadcast approximately.
Here are my settings:
[image]
[image]
Is there anything wrong with this?

JsBergbau, 2021-12-18:
Very strange issue. I've updated one Mi LYWSD03MMC to the latest version and pulled the repo. It works with `./LYWSD03MMC.py --atc`

Did you send the settings and thermometer stayed connected afterwards? That is important that settings get applied. After that you have to disconnect from the Telink-Flasher, Bluetooth sign must vanish from display, only then data is broadcasted.

SydneyOwl, 2021-12-18:
Thank you for your reply!
I think settings are properly storaged. I also tried removing pin and flashing the firmware again but they did not work.
So the problem may be ubuntu20.04 and I'll try to switch to raspbian and run it on raspberry zero w later.

SydneyOwl, 2022-01-14:
Works well on raspberry pi zero w. Problem solved.

closed 2022-01-14 by SydneyOwl