JsBergbau/MiTemperature2 #107: Negative temperature in atc mode

merluxx, 2021-12-22:
Hello and first of all thank you for your precious code. 
I noticed an error when the thermotres returns a negative temperature and when the script is in ATC mode. 
I was able to correct it by modifying line 463 of LYWSD03MMC.py it is necessary to replace: 
temperature = int (data_str [22:26], 16) / 10. by 
temperature = int.from_bytes (data [22:26], byteorder = 'little', signed = True) / 10.
Could you integrate it?
Cordially.

merluxx, 2021-12-23:
with Atc1441 as Advertising type

JsBergbau, 2021-12-23:
Hallo @merluxx,

thanks for appreciating MitTemperature2 software and also improving it.

What version of MiTemperature2 do you have? If it doesn't print a version, when starting the script, then it is quite old. There was indeed an issue with negative temperature and I think I fixed this about a year ago. Line 463 is now `connectionLostCounter=0`.

merluxx, 2021-12-23:
indeed you are right, I had an old version. Everything works with the latest release. Thank you

closed 2021-12-23 by merluxx