JsBergbau/MiTemperature2 #110: Send to File no data

Volverine, 2022-01-13:
My scripts kinda working but in the file its just reporting somthing like this:
```sensorname,temperature,humidity,voltage,batteryLevel,timestamp``` instead of real values which are showing properly in the console.

What i'm doing wrong?

JsBergbau, 2022-01-13:
I've just tested and it works as expected.
To debug this:
Please provide your full command line starting MitTemperature2 and the output of the script. 
Please also note that it takes a few seconds until data is written. So please let the script run a few seconds after receiving the first data.

Volverine, 2022-01-13:
```/usr/bin/python3 /home/pi/domoticz/scripts/python/MiTemperature2/LYWSD03MMC.py -d <MAC> -r -deb -b -c 1 -call ./sendToFile.sh```

And this is how it looks in the console:
```Trying to connect to MAC
Connection lost
Waiting...
Trying to connect to MAC
Temperature: 5.4
Humidity: 82
Battery voltage: 2.97 V
Battery level: 87
1 measurements collected. Exiting in a moment.
/home/pi/domoticz/scripts/python/MiTemperature2/./sendToFile.sh sensorname,temperature,humidity,voltage,batteryLevel,timestamp MAC 5.4 81 2.97 87 1642095090
```

Volverine, 2022-01-13:
nvm. its working now.

closed 2022-01-13 by Volverine