JsBergbau/MiTemperature2 #111: sendToDomoticz gives no result

zozolpl, 2022-01-20:
Hello, Im trying to retrieve values on domoticz, but it is not working. 
I added Dummy in domoticz, created Virtual Device as Temp+Hum, it gets IDX 19 
Then I eddited sendToDomoticz.py to 
domoticzserver = "127.0.0.1:8080" #script work on the same raspbery where domoticz stands
idx_temp = "19"

When I run  $ /usr/bin/python3 ~/domoticz/scripts/python/MiTemperature2/LYWSD03MMC.py -d A4:C1:38:B8:CF:1C -r -b -c 1 -call ./sendToDomoticz.py output:

`---------------------------------------------
MiTemperature2 / ATC Thermometer version 5.0
---------------------------------------------

Please read README.md in this folder. Latest version is available at https://github.com/JsBergbau/MiTemperature2#readme
This file explains very detailed about the usage and covers everything you need to know as user.

Trying to connect to A4:C1:38:B8:CF:1C
Temperature: 24.6
Humidity: 47
Battery voltage: 3.262 V
Battery level: 100
1 measurements collected. Exiting in a moment.
/home/pi/domoticz/scripts/python/MiTemperature2/./sendToDomoticz_19.py sensorname,temperature,humidity,voltage,batteryLevel,timestamp A4:C1:38:B8:CF:1C 24.6 47 3.262 100 1642682285
`

So everything looks good, but in Domoticz Im not reading values - it still shows 0C and 50% humidity

findryDev, 2022-01-20:
Do you have a password-protected domoticz?

zozolpl, 2022-01-20:
Yes, but Ive added my local networks IP to disable passwords

findryDev, 2022-01-20:
I used IP: "127.0.0.1;192.168.0.*" in password disable network address.

zozolpl, 2022-01-20:
I found the bug - I had a space after first ; and the others IP didn't work as a local area network