JsBergbau/MiTemperature2 #114: Howto use  options --devicelistfile and --onlydevicelist

TaiJiGong, 2022-02-01:
I can't use --devicelistfile and --onlydevicelist, there are no results. The script output is only:

`
MiTemperature2# ./LYWSD03MMC.py --watchdogtimer 5 --unreachable-count 5 --passive --devicelistfile sensors-test.ini --onlydevicelist
---------------------------------------------
MiTemperature2 / ATC Thermometer version 5.0
---------------------------------------------

Please read README.md in this folder. Latest version is available at https://github.com/JsBergbau/MiTemperature2#readme
This file explains very detailed about the usage and covers everything you need to know as user.

Script started in passive mode
------------------------------
Power ON bluetooth device 0
Bluetooth device 0 is already enabled
Enable LE scan
scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no
socket filter set to ptype=HCI_EVENT_PKT event=LE_META_EVENT
Listening ...

`

> sensors-test.ini file:

`
[A4:C1:38:15:3A:98]
sensorname = Sensor 4 15:3A:98

[A4:C1:38:0E:E2:7B]
sensorname = Sensor 2 0E:E2:7B

[A4:C1:38:6E:CF:F0]
sensorname = Sensor 5 6E:CF:F0`

With option -d, the script is working well:
`
MiTemperature2# ./LYWSD03MMC.py --watchdogtimer 5 --unreachable-count 5 --passive -d A4:C1:38:6E:CF:F0
---------------------------------------------
MiTemperature2 / ATC Thermometer version 5.0
---------------------------------------------

Please read README.md in this folder. Latest version is available at https://github.com/JsBergbau/MiTemperature2#readme
This file explains very detailed about the usage and covers everything you need to know as user.

Trying to connect to A4:C1:38:6E:CF:F0
Temperature: 18.84
Humidity: 49
Battery voltage: 2.45 V
Battery level: 35

Temperature: 18.81
Humidity: 49
Battery voltage: 2.45 V
Battery level: 35

`

JsBergbau, 2022-02-01:
Hallo TaiJiGong,

with `-d` option you connect to a single device. This mode is not recommended any more and will be dropped in future versions, because it drains battery quite fast. You have to open the program for every sensor with `-d` option you want to read.

In passive mode the script listens for all sensors around and so there is the option to show only values from desired sensors. With passive mode `--unreachable-count` makes no sense, because this option is only when a sensor can't be connected to.

When you want to use passive mode, you have to flash your sensors with ATC firmware https://github.com/pvvx/ATC_MiThermometer

TaiJiGong, 2022-02-01:
Hallo JsBergbau,

Thanks for your replay. After flashing the ATC firmware it works well.

closed 2022-02-01 by TaiJiGong