JsBergbau/MiTemperature2 #116: Feature request: Only trigger callback every n minutes

fwieringen, 2022-02-14:
Hi,

Impressed with the script. Took me less than a minute to get the data out of the temperature sensor.

I'm running Domoticz on a RPi Zero W. I notice that having the script trigger the callback function every reading will cause the CPU usage to come from 5% to about 60%.

It's quite acceptable to have the temp sensor send an update to Domoticz every 10 minutes instead of every 6 seconds. I now do this by scheduling a cronjob and set the -c 1 parameter of the script. 

However, it would be nice when the script can be configured with a command line parameter to only send updates every n minutes.

Thanks!

Cheers, Friso

JsBergbau, 2022-02-14:
Hi fwieringen,

thanks for your praise.

I had experienced a quite similiar issue and so an integrated MQTT client was added because starting a new process is quite CPU intensive.

Using `-c 1` parameter implies that you're using connected mode. Note: This mode is deprecated and will be removed in a future version. You should have a look at passive mode https://github.com/JsBergbau/MiTemperature2#passive-mode-usage , since this has a significant longer battery life.
Concerning your request: There is the `--skipidentical N` parameter. So if your sensor reports the same values it will not be send to callback. That is quite similiar what you're looking for and maybe even better. If there is a new measurement every 6 seconds, you set it to 10 * 60 / 6 = 100. So about every 10 minutes callback will be executed but if temperature and/or humidity changes it will be reported as soon as it changes.

fwieringen, 2022-02-15:
Hi JSBergbau,

Thanks for your answer. I didn't look into the passive mode yet. Flashed the firmware and now this works.

Considering the skipidentifical. I tried this, but the temp sensor or humidity sensor changes quites regular, sometimes just a decimal. This caused the script to still send updates every 12 to 18 seconds.

I had a quick look at the code, but I think I can implement a callback limit with an additional command line parameter without to many effort. 

Thanks, Friso

JsBergbau, 2022-02-15:
> Considering the skipidentifical. I tried this, but the temp sensor or humidity sensor changes quite regular, sometimes just a decimal. This caused the script to still send updates every 12 to 18 seconds.

Therefore is a rounding option to mitigate that problem.

fwieringen, 2022-02-16:
Closed by #118

closed 2022-02-16 by fwieringen