JsBergbau/MiTemperature2 #120: Fehlermeldung humidity

RingoErdy, 2022-02-18:
Hallo wenn ich das 
das semikolon vor humidityOffset=-5 entferne habe ich immer folgende Fehlermeldung
*
*
*
BLE packet - Custom: A4:C1:38:22:F5:54 00 1312161a1854f52238c1a4e9085211e90a41f804 -76
Temperature:  22.8
Humidity:  44.3
Battery voltage: 2.793 V
RSSI: -76 dBm
Battery: 65 %
Exception when calling handler with a BLE advertising event: NameError("name 'humidity' is not defined")
Traceback (most recent call last):
  File "/home/openhabian/MiTemperature2/bluetooth_utils.py", line 372, in parse_le_advertising_events
    handler(mac_addr_str, adv_type, data, rssi)
  File "LYWSD03MMC.py", line 758, in le_advertise_packet_handler
    measurement.humidity = humidity + int(sensors[mac]["humidityOffset"])
NameError: name 'humidity' is not defined
*
*
*
wenn ich es vor ;offset1 = -2 und ;offset2 = 2 entferne habe ich
*
*
*
BLE packet - Custom: A4:C1:38:22:F5:54 00 1312161a1854f52238c1a4f7084211ef0a420b04 -68
Temperature:  22.9
Humidity:  44.2
Battery voltage: 2.799 V
RSSI: -68 dBm
Battery: 66 %
Exception when calling handler with a BLE advertising event: NameError("name 'humidity' is not defined")
Traceback (most recent call last):
  File "/home/openhabian/MiTemperature2/bluetooth_utils.py", line 372, in parse_le_advertising_events
    handler(mac_addr_str, adv_type, data, rssi)
  File "LYWSD03MMC.py", line 755, in le_advertise_packet_handler
    measurement.humidity = calibrateHumidity2Points(humidity,int(sensors[mac]["offset1"]),int(sensors[mac]["offset2"]),int(sensors[mac]["calpoint1"]),int(sensors[mac]["calpoint2"]))
NameError: name 'humidity' is not defined
*
*
*
[A4:C1:38:22:F5:54]
sensorname=Wohnzimmer
;humidityOffset=-5
;offset1 = -2
;offset2 = 2
calpoint1 = 33
calpoint2 = 75
topic=basement/Wohnzimmer

Danke Ringo

JsBergbau, 2022-02-19:
Hallo Ringo,

danke für die Meldung. Es war tatsächlich ein Bug im Skript. Dieser ist nun behoben, zusätzlich wird jetzt der Sensorname angezeigt. Probier es mal aus und sag bescheid, ob jetzt alles zufriedenstellend funktioniert.

VG,
JsBergbau

RingoErdy, 2022-02-19:
Ok, bin gespannt.

Schönes WE

RingoErdy, 2022-02-21:
Problem gelöst