JsBergbau/MiTemperature2 #126: _bluetooth.error: (100, 'Network is down')

stefanheijnen, 2022-07-10:
```
Jul 10 08:58:06 pi0-slaapkamer LYWSD03MMC.py[26221]: Exception in thread Thread-2:
Jul 10 08:58:06 pi0-slaapkamer LYWSD03MMC.py[26221]: Traceback (most recent call last):
Jul 10 08:58:06 pi0-slaapkamer LYWSD03MMC.py[26221]:   File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
Jul 10 08:58:06 pi0-slaapkamer LYWSD03MMC.py[26221]:     self.run()
Jul 10 08:58:06 pi0-slaapkamer LYWSD03MMC.py[26221]:   File "/usr/lib/python3.7/threading.py", line 865, in run
Jul 10 08:58:06 pi0-slaapkamer LYWSD03MMC.py[26221]:     self._target(*self._args, **self._kwargs)
Jul 10 08:58:06 pi0-slaapkamer LYWSD03MMC.py[26221]:   File "/opt/MiTemperature2/LYWSD03MMC.py", line 197, in keepingLEScanRunning
Jul 10 08:58:06 pi0-slaapkamer LYWSD03MMC.py[26221]:     disable_le_scan(sock)
Jul 10 08:58:06 pi0-slaapkamer LYWSD03MMC.py[26221]:   File "/opt/MiTemperature2/bluetooth_utils.py", line 230, in disable_le_scan
Jul 10 08:58:06 pi0-slaapkamer LYWSD03MMC.py[26221]:     bluez.hci_send_cmd(sock, OGF_LE_CTL, OCF_LE_SET_SCAN_ENABLE, cmd_pkt)
Jul 10 08:58:06 pi0-slaapkamer LYWSD03MMC.py[26221]: _bluetooth.error: (100, 'Network is down')
```

I get this error up to a few times a day.

Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster

Raspberry Pi Zero W Rev 1.1

Running with: `ExecStart=/opt/MiTemperature2/LYWSD03MMC.py --mqttconfigfile /opt/MiTemperature2/mqtt.conf --atc --watchdogtimer 5 --devicelistfile /opt/MiTemperature2/sensors.ini --onlydevicelist`

For now I work around this by restarting the service every 4 hours.

Is it possible to catch this exception in your code?

JsBergbau, 2022-07-10:
Hallo stefanheijnen,

I'm running the code on 3 raspberry PIs Zero W devices and never got that error. 
Do you have any other programs accessing bluetooth? It looks like this. Error network is down normaly occurs when doing `sudo hciconfig hci0 down`.
So having this error means there is definitely something wrong.

stefanheijnen, 2022-07-10:
Hallo JsBergbau,

I'd also have https://github.com/MartinHaimberger/miflora-mqtt-daemon running for the older, round, Xiaomi temperature/humidity sensors (MJ_HT_V1).

JsBergbau, 2022-07-10:
I've made the experience, that systems that only run MiTemperature 2 would even run without the watchdog flawlessly. As soon as there is something else on the system using bluetooth watchdog gets more important, because doing something else on bluetooth interface often disables LE-Scanning. However I've never encountered that hci interface went down.

For me this looks like a bug in miflora deamon because that software should never disable hci interface. 

Is that daemon running with root rights? A normal user shouldn't be able to bring hci interface down.

stefanheijnen, 2022-07-12:
Seems you are right:

```
pi@pi0-slaapkamer:~ $ sudo systemctl status miflora.service 
● miflora.service - Xiaomi Mi Flora Plant Sensor MQTT Client/Daemon
   Loaded: loaded (/etc/systemd/system/miflora.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2022-07-05 10:54:57 CEST; 1 weeks 0 days ago
     Docs: https://github.com/ThomDietrich/miflora-mqtt-daemon
 Main PID: 447 (python3)
   Status: "Jul 12 16:00:24 - Failed to retrieve data from Mijia Bluetooth Temperature Smart Humidity sensor "Badkamer" (4C:65:A8:DE:FE:72), success rate: 100%."
    Tasks: 4 (limit: 877)
   CGroup: /system.slice/miflora.service
           └─447 python3 /opt/miflora-mqtt-daemon/miflora-mqtt-daemon.py

Jul 12 16:46:26 pi0-slaapkamer sudo[16011]: pam_unix(sudo:session): session closed for user root
Jul 12 16:47:48 pi0-slaapkamer sudo[16019]:       pi : TTY=unknown ; PWD=/opt/miflora-mqtt-daemon ; USER=root ; COMMAND=/bin/hciconfig hci0 reset
Jul 12 16:47:48 pi0-slaapkamer sudo[16019]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jul 12 16:47:49 pi0-slaapkamer sudo[16019]: pam_unix(sudo:session): session closed for user root
Jul 12 16:49:08 pi0-slaapkamer sudo[16028]:       pi : TTY=unknown ; PWD=/opt/miflora-mqtt-daemon ; USER=root ; COMMAND=/bin/hciconfig hci0 reset
Jul 12 16:49:08 pi0-slaapkamer sudo[16028]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jul 12 16:49:09 pi0-slaapkamer sudo[16028]: pam_unix(sudo:session): session closed for user root
Jul 12 16:50:32 pi0-slaapkamer sudo[16068]:       pi : TTY=unknown ; PWD=/opt/miflora-mqtt-daemon ; USER=root ; COMMAND=/bin/hciconfig hci0 reset
Jul 12 16:50:32 pi0-slaapkamer sudo[16068]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jul 12 16:50:33 pi0-slaapkamer sudo[16068]: pam_unix(sudo:session): session closed for user root
```

Let's see if I can disable this behavior.

Thanks for pointing me in the right direction!

closed 2022-07-12 by stefanheijnen