JsBergbau/MiTemperature2 #127: Connection lost loop

pristis, 2022-08-29:
Hello!

I have an issue with the script. I am using this command to get data from my temp and hum sensor:

`python3.9 LYWSD03MMC.py -d A4:C1:38:B4:48:55 -r -b`

And the answer from console is always this loop:
```
Trying to connect to A4:C1:38:B4:48:55
Connection lost
Waiting...
Trying to connect to A4:C1:38:B4:48:55
Connection lost
Waiting...
```
I can connect with `gatttool -I` with no issues, but with LYWSD03MMC.py I can't find a solution.

I will be glad for some help. Thanks for your time.

JsBergbau, 2022-08-30:
Hi pristis,

which Firmware are you using? If not already ATC Firmware von there https://github.com/pvvx/ATC_MiThermometer I recommend using this Firmware. Connection mode is not recommended any more and support will be removed in a future version.

wojtekza, 2023-10-27:
hello, while trying to connect the sensor to domoticz, I updated the sensor.  When trying to change the software, I get the message 
10:16:48: Searching for devices
 10:16:54: Connecting to: LYWSD03MMC
 10:16:58: Hardware Revision String: B1.5
 10:16:58: Software Revision String: 0159
 10:16:58: Firmware Revision String: 2.1.1_0159
 10:16:58: Detected Mi device
 10:16:58: (!) Version: 2.1.1_0159 is temporarily not supported (!)
 10:16:58: Disconnected.  

Will he be able to change the software version in some way?

plontajos, 2024-12-16:
I have strange issue, 5 sensors works perfectly (3 pcs on RPI4, 2 pcs on RPI0).
Bought next 4 and I'm trying to connect it without luck.

**The diffrence is in value coding 0x0033 for new one vs 0x0036 for old one.**

```
Trying to connect to A4:C1:38:58:D9:2C
Connection lost
```

I can connect the gattool

Non-working sensor (new one)
```
Attempting to connect to A4:C1:38:58:D9:2C
Connection successful
Notification handle = **0x0033** value: 29 07 3d 03 0c 
```
Normally working sensor (old one)
```
Attempting to connect to A4:C1:38:C9:13:2C
Connection successful
Notification handle = **0x0036** value: 85 07 41 c8 0b 
```
Another normally working sensor (old one)
```
Attempting to connect to A4:C1:38:8F:16:D8
Connection successful
Notification handle = **0x0036** value: 82 07 3d 93 0b 
```