JsBergbau/MiTemperature2 #129: Add image to dockerhub

antxxxx, 2022-09-06:
Would it be possible to add the docker image for this to dockerhub to make it easier to install and run?   

I am not sure if you want to create an account on dockerhub to host this (I can provide instructions for building this for multiple architectures and pushing it if you want), or if you are happy to have it under my dockerhub account

JsBergbau, 2022-09-06:
Since I don't have a dockerhub account and re-building for new versions takes quite some time, it is ok if you host it under your dockerhub. Please add link to this repository so in case dockerhub images get outdated the latest version can be found here.

antxxxx, 2022-09-06:
Image is available at https://hub.docker.com/repository/docker/antxxxx/mitemperature2/. Its currently just tagged as latest as there seem to be a few changes since the v5 release and I wanted to use the smaller docker image that you merged earlier

I have also created https://github.com/JsBergbau/MiTemperature2/pull/130 with instructions on building and pushing to dockerhub and how to use it on a raspberry pi in the main readme