JsBergbau/MiTemperature2 #133: Passive and callback ?

Poil, 2022-10-11:
Hi,

I'm trying to switch from "normal mode" to passive, but the callback script is only called once at first poll, what I am missing ?
(Also it should be nice to have an option to hide the message)

```
 ./LYWSD03MMC.py --passive --atc --watchdogtimer 5 --devicelistfile mysensors.ini --onlydevicelist --callback ./sendToPrometheusPushGateway.sh # --callback-interval 60 ## also tried with this
```

Output :
```
$ python3 ./LYWSD03MMC.py  --passive --atc --watchdogtimer 5 --devicelistfile mysensors.ini --onlydevicelist --callback ./sendToPrometheusPushGateway.sh --callback-interval 60
---------------------------------------------
MiTemperature2 / ATC Thermometer version 5.0
---------------------------------------------

Please read README.md in this folder. Latest version is available at https://github.com/JsBergbau/MiTemperature2#readme
This file explains very detailed about the usage and covers everything you need to know as user.

Script started in passive mode
------------------------------
In this mode all devices within reach are read out, unless a devicelistfile and --onlydevicelist is specified.
Also --name Argument is ignored, if you require names, please use --devicelistfile.
In this mode debouncing is not available. Rounding option will round humidity and temperature to one decimal place.
Passive mode usually requires root rights. If you want to use it with normal user rights, 
please execute "sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f `which python3`)"
You have to redo this step if you upgrade your python version.
----------------------------
Power ON bluetooth device 0
Bluetooth device 0 is already enabled
Enable LE scan
scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no
socket filter set to ptype=HCI_EVENT_PKT event=LE_META_EVENT
Listening ...
BLE packet - Custom: A4:C1:38:FE:BB:BB 00 1312161a18bbbbfe38c1a41007e31bf20926180f -43
Sensorname: cave_cellier
Temperature:  18.08
Humidity:  71.39
Battery voltage: 2.546 V
RSSI: -43 dBm
Battery: 38 %

/home/benjamin/pi/MiTemperature2/./sendToPrometheusPushGateway.sh sensorname,temperature,humidity,voltage,timestamp cave_cellier 18.08 71.39 2.546 1665481923
BLE packet - Custom: A4:C1:38:BB:54:29 00 1312161a182954bb38c1a4bf078e1b130b46da0f -77
Sensorname: cave_bureau
Temperature:  19.83
Humidity:  70.54
Battery voltage: 2.835 V
RSSI: -77 dBm
Battery: 70 %

---------------------------------------------
MiTemperature2 / ATC Thermometer version 5.0
---------------------------------------------

Please read README.md in this folder. Latest version is available at https://github.com/JsBergbau/MiTemperature2#readme
This file explains very detailed about the usage and covers everything you need to know as user.

BLE packet - Custom: A4:C1:38:1E:8E:43 00 1312161a18438e1e38c1a45a07fd191b0b47de0f -78
Sensorname: escalier_sous_sol
Temperature:  18.82
Humidity:  66.53
Battery voltage: 2.843 V
RSSI: -78 dBm
Battery: 71 %

Trying to connect to A4:C1:38:0B:F8:E1
Connection lost
Maximum numbers of unsuccessful connections reached, exiting
---------------------------------------------
MiTemperature2 / ATC Thermometer version 5.0
---------------------------------------------
```

JsBergbau, 2022-10-11:
Can you please provide the output over a longer time period? 

```
Callback for Sensor1 would be within interval (55 < 60); don't invoke callback
```
You should see messages of this kind. 

To suppress output append ` >/dev/null` after your command.

Poil, 2022-10-11:
My bad ...

It just I was calling the wrong callback script !

Thanks

closed 2022-10-11 by Poil