JsBergbau/MiTemperature2 #134: Python 3.5 version support

alex3594, 2022-11-07:
Hello! 
Can you please create the latest script version for the Python3.5? 
I have an embedded controller where I want to run your script, but it cant be upgraded to the Python 3.7
Thanks in advance!

alex3594, 2022-11-07:
Here is the error I see with your old script version:
```
python3 LYWSD03MMC.py --device A4:C1:38:C7:45:3B
Trying to connect to A4:C1:38:C7:45:3B
Traceback (most recent call last):
  File "LYWSD03MMC.py", line 111, in <module>
    if p.waitForNotifications(2000):
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 560, in waitForNotifications
    resp = self._getResp(['ntfy','ind'], timeout)
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 407, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 338, in _waitResp
    if self._helper.poll() is not None:
AttributeError: 'NoneType' object has no attribute 'poll'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "LYWSD03MMC.py", line 124, in <module>
    p=connect()
  File "LYWSD03MMC.py", line 56, in connect
    p = btle.Peripheral(adress)
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 391, in __init__
    self._connect(deviceAddr, addrType, iface)
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 439, in _connect
    raise BTLEDisconnectError("Failed to connect to peripheral %s, addr type: %s" % (addr, addrType), rsp)
bluepy.btle.BTLEDisconnectError: Failed to connect to peripheral A4:C1:38:C7:45:3B, addr type: public

```

JsBergbau, 2022-11-07:
Hallo alex3594,

rewriting for Python 3.5 is some effort. Python 3.5 has reached end of life more than 2 years ago (13 Sep 2020) https://endoflife.date/python
I don't find it useful to rewrite software to support python version which has reached end of life, sorry.

closed 2022-11-07 by JsBergbau

alex3594, 2022-11-07:
I see, thanks for the update. 
But can you at least fix the bug you mentioned in your version? https://raw.githubusercontent.com/JsBergbau/MiTemperature2/5d7b215d7b22d4c21d9244f8a4102513b928f2c7/LYWSD03MMC.py

JsBergbau, 2022-11-07:
That script is very old. It only supports connection mode, which is deprecated now in favor of passive mode. Code for connection mode will be removed in a future version. So sorry, also no further development there.

alex3594, 2022-11-07:
Got it. Will try to upgrade controller to the python3.7