JsBergbau/MiTemperature2 #136: Domoticz issue after update 2023.1

ymmi21, 2023-02-19:
Hi all,
That's my first post here but I can't find any solution for my issue. 
So I would like to describe my issue:
I received an update for Domoticz, so I made it. But after update, system wants my login and pass (I forgot it in the meantime) so I need to reset a pass. 
After logging into the dashboard all my sensors disappeared from there. I found a solution for it (enable API protection), sensors back (uff). But they stop count a siganls since yesterday evening and I don't know why.

I tried a lot of options but without a success :(
I found a strange thing:
when I try to use a command "_LYWSD03MMC.py -d a4:c1:38:fe:4f:7b -r -deb -b -c 1 -call ./sendToDomoticz1.py_" I receive "_**command not found**_"
[image]

but after I add a "python3 ./" before command averything pass, like this:
[image]

but at the end is "_**Data couldn't be send to Callback, etc.**_"

Please help me repair it. Thx all for all

ymmi21, 2023-02-19:
Ok, it's working :)
I add an option in security tab:
[image]

mafrei2019, 2024-08-31:
Trying to connect to A4:C1:38:xx:xx:xx
Connection lost
wie sieht deine sendToDomoticz.py aus ?