JsBergbau/MiTemperature2 #137: sensor.ini

mafrei2019, 2023-03-05:
Hi, ich habe das ganze nun mal auf Docker installiert. im Status steht immer created.
Wo muss den die Sensor.ini stehn damit die Geräte gefunden werden.
Seit dem Update von Python läuft es nicht mehr, deshalb test mit Docker.

JsBergbau, 2023-03-05:
Welcher Fehler kommt denn seit dem Pythonupdate?
Hast du schon versucht `sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f `which python3`)` auszuführen? 

Mit Docker selbst kann ich dir leider nicht weiterhelfen.

mafrei2019, 2023-03-05:
Ja hatte ich gemacht, liegt wohl an Bluetooh, er sagt zwar aktiv kann aber den Controller nicht finden.
Script started in ATC Mode
----------------------------
In this mode all devices within reach are read out, unless a devicelistfile and --onlydevicelist is specified.
Also --name Argument is ignored, if you require names, please use --devicelistfile.
In this mode debouncing is not available. Rounding option will round humidity and temperature to one decimal place.
ATC mode usually requires root rights. If you want to use it with normal user rights,
please execute "sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f `which python3`)"
You have to redo this step if you upgrade your python version.
----------------------------
Power ON bluetooth device 0
Traceback (most recent call last):
  File "/home/pi/./MiTemperature2-4.0/LYWSD03MMC.py", line 586, in <module>
    toggle_device(dev_id, True)
  File "/home/pi/MiTemperature2-4.0/bluetooth_utils.py", line 122, in toggle_device
    fcntl.ioctl(hci_sock.fileno(),
OSError: [Errno 19] No such device

JsBergbau, 2023-03-05:
Die Meldung ist mit Docker oder ohne?

Was kommt denn bei `hciconfig -a` (ohne Dockercontainer)?

mafrei2019, 2023-03-05:
> Ohne Docker (läuft ja auch nicht)

hciconfig -a 
keine Reaktion

auf  sudo systemctl status hciuart.service....kommt
● hciuart.service - Configure Bluetooth Modems connected by UART
     Loaded: loaded (/lib/systemd/system/hciuart.service; enabled; vendor prese>
     Active: inactive (dead)
  Condition: start condition failed at Sun 2023-03-05 12:55:36 CET; 7h ago
             └─ ConditionFileNotEmpty=/proc/device-tree/soc/gpio@7e200000/bt_pi>

>

JsBergbau, 2023-03-05:
Das ist definitiv ein Problem mit dem System, da er bei dir kein Bluetoothgerät findet. Ich würde die ganze Hardware mal für ein paar Minuten stromlos machen und dann schauen, dass bei `hciconfig -a` etwas angezeigt wird, ansonsten wirst du nicht weiterkommen.

mafrei2019, 2023-03-06:
Warum auch immer war in der  sudo nano /boot/config.txt Datei Bluetooth deaktiviert.
 Nun läuft es wieder (ohne Docker).
Wenn jemand Tipps für Docker Einbindung hat, bitte posten.

closed 2023-03-06 by mafrei2019