JsBergbau/MiTemperature2 #139: Enable BT5 Long Range

Poil, 2023-03-28:
Hi,

Is it possible to enable BT5 PHY/LE ? I think we have to patch bluetooth_utils.py but I'm not sure ?

It looks like we need to configure the adaptator to

```
hcitool -i hciX cmd 08 31 03 05 05
# or 
hcitool -i hciX cmd 08 31 03 07 07
```

```
< HCI Command: LE Set Default PHY (0x08|0x0031) plen 3                                                                                                                                                                                                                                                   #17 [hci1] 6.052660
        All PHYs preference: 0x03
          No TX PHY preference
          No RX PHY preference
        TX PHYs preference: 0x05
          LE 1M
          LE Coded
        RX PHYs preference: 0x05
          LE 1M
          LE Coded
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                                                                                                                              #18 [hci1] 6.053720
      LE Set Default PHY (0x08|0x0031) ncmd 2
        Status: Success (0x00)
```

ref. https://github.com/custom-components/ble_monitor/issues/1135 & https://github.com/Bluetooth-Devices/bthome-ble/issues/45

Requirements

```
# cat /etc/systemd/system/bluetooth.service.d/override.conf 
[Service]
Type=dbus
BusName=org.bluez
ExecStart=
ExecStart=/usr/lib/bluetooth/bluetoothd --experimental

# bluetoothd  --version
5.66
```

Best regards

JsBergbau, 2023-03-28:
That looks like a systemwide change. I'm not sure whether this should be done at all in MiTemperature2.
What is your goal? Since we are only passive listening, it shouldn't matter whether Bluetooth5 is enabled or not.

Poil, 2023-03-28:
BT5 is OK, but not I'm unable to find any sensor when I enable Long Range on firmware 4.2 (https://github.com/pvvx/ATC_MiThermometer)
I thank you override some adaptator configuration, but perhaps I'm wrong

JsBergbau, 2023-03-28:
Thanks for the link. Now I understand.

Raspberry PIs seem not to support BT5 Long range mode https://forums.raspberrypi.com/viewtopic.php?t=244386#p1523836

> So the only thing they seem to have done is a new qualification for the latest version, and not actually implemented any of the new optional features such as 2 Mbit/s PHY, Long Range, Extended advertisements etc.

So when trying this with raspberry PI, you won't see the sensors, even with "BT5" enabled.

Poil, 2023-03-28:
I'm trying on a X86 machine with a BT5.1 USB dongle :)

JsBergbau, 2025-04-28:
Hi @Poil,

sorry for the late reply. I started using Bluetooth 5.0 long range a few days ago. There is no need to activate anything with ZEXMTE 5.4 USB Bluetooth Adapter https://www.amazon.de/ZEXMTE-Bluetooth-Adapter-Wireless-Receiver-Schwarz-Gold/dp/B09NVPTS6J

It works out of the Box with MiTemperature2 Script.

closed 2025-04-30 by JsBergbau