JsBergbau/MiTemperature2 #145: Update firmaware to 2.1.1_0159

arko85, 2023-11-19:
After udpdate can not connect with LYWSD03MMC

JsBergbau, 2023-11-20:
How do you try to connect?

arko85, 2023-11-21:
I use :
`/home/pi/scripts/python/MiTemperature2/LYWSD03MMC.py -d A4:C1:38:BE:6C:A8 -r -deb -b -c 1 -urc 10 -call ./sendToDomoticz5.py`

JsBergbau, 2023-11-21:
Connection mode is deprecated and will be removed in a future version. Probably it has been removed in the latest firmware. Please use passive mode
https://github.com/JsBergbau/MiTemperature2#passive-mode-usage

arko85, 2023-11-21:
To use passive mode I need use alternative firmware from atc1441 or pvvx?

JsBergbau, 2023-11-21:
Take the firmware from pvvx

arko85, 2023-11-21:
firmware 2.1.1_0159 is not supported by pvvx

arko85, 2023-11-21:
it look like there are no  problem solution for firmware 2.1.1_0159